AmboVent-1690-108/AmboVent #2: Post 3D print model file to 3D print Covid-19 group?

carolynnicita, 2020-04-02:
Could you please notify the Covid-19 3D printing group where we could find the model file(s) of the 3D printed parts for your design?  Thanks much!

the Discord room
https://discord.gg/GCTW4G

nimrod46, 2020-04-02:
Could you check if everything you need is [here](https://github.com/AmboVent/AmboVent/tree/master/Mechanical%20Design)? we just added the mechanical designs

KKaldis, 2020-04-02:
there are  *.step files, you can output with any software to stl

jejaffeTN, 2020-04-02:
Are CAD files available in other formats such as **DAE, OBJ or STL?** 

If not, and we convert them, may we post them here for others? 

Jonathan

closed 2020-04-02 by carolynnicita

jejaffeTN, 2020-04-02:
Closed without comment or answer by CarolynNicita
I do not understand.

salmanfarisvp, 2020-04-08:
- SolidWorks Files: [SolidWorks Pack and Go](https://github.com/AmboVent/AmboVent/tree/master/.Mechanics/Mechanism/SolidWorks%20Pack%20and%20Go)

- [Assembly in other formats](https://github.com/AmboVent/AmboVent/tree/master/.Mechanics/Mechanism/Assembly%20in%20other%20formats)