AmboVent-1690-108/AmboVent #9: RST does something random

airbornemint, 2020-04-05:
This line should be `digitalRead(PIN_RST)`

https://github.com/AmboVent/AmboVent/blob/e889eb9853d8ce3cfa903381849227c7397c351b/.Software/Arduino%20Code/ventilation_machine/ventilation_machine.ino#L502

salam17, 2020-04-05:
RST is initialized to zero so the first digitalRead(RST) call is equivalent to digitalRead(pin_POT). On the next call. If the digitalRead returned zero, RST becomes 1 and the next read will be from pin_CUR, not an expected result.  RST is used to set the sign of wanted_cal_PWM so reading the rough position of pin_POT makes sense ( 0 for a position close to the minimum).  Should line 502 be:
RST = (1-digitalRead (pin_POT));  ?

nimrod46, 2020-04-06:
I fixed this line but keep in mind that this button is not currently in use, it's used only in calibration

closed 2020-04-06 by nimrod46