AmboVent-1690-108/AmboVent #15: Encourage community involvement

maorfr, 2020-04-08:
Hello folks,

First and foremost - THANK YOU!!

As this repository is open source, it will be amazing to encourage the open source community to get involved in the work being done.

A good way to get help is to create issues stating what help is needed. To each such issue, you can add the `help-wanted` label.

There are many who would like to help! Help us help you!

Thank you again. This project is TRULLY a life saver.

MatanYemini, 2020-04-08:
Waiting for you to add more issues, or just share what is planned:)

nimrod46, 2020-04-12:
Thank you! We will keep that in mind, for now, I will close the issue.
Feel free to open PR as you feel the need to improve things

closed 2020-04-12 by nimrod46