AmboVent-1690-108/AmboVent #16: Solidworks files are not portable

mgaman, 2020-04-10:
For those of us working at home with open source CAD software, the SLDPRT and SLDASM files are unusable. 
Can you please export into some compatible format e.g. STEP.

SRC0991, 2020-04-10:
Hi, my friend

You can do two things for work with this files.
1. You can download Drawings to open SLDPRT and SLDASM files.
Here is the link: https://www.edrawingsviewer.com/download-edrawings
2. You can convert to CAD software such as STEP using CAD exchanger 
Here is the link: https://cadexchanger.com/downloads

mgaman, 2020-04-11:
Thanks. I tried cadexchanger and was able to import some SLDPRT and export in brep and stl format.
SLDDRW files are ignored, SLDASM sometimes yes, sometimes no.
There is a sizing issue e.g. Cadexchanger imports a SLDPRT file and the dimension tool shows it to be 206mm long. However I export the part as SfTL and open with CURA and the part seems to be >300mm long.
I assume that I'm messing up somewhere so I'll have to go back and RTFM.

justinclift, 2020-04-12:
As a data point, there are STEP format versions of some of the files in there.  eg:

* https://github.com/AmboVent/AmboVent/blob/a48090222f2ed3abc7753daacaff11675f97fa73/.Mechanics/Casing/AMBOVENT.STEP
* https://github.com/AmboVent/AmboVent/blob/a48090222f2ed3abc7753daacaff11675f97fa73/.Mechanics/Mechanism/Assembly%20in%20other%20formats/1690-20-8300%20%20REV7%20%2026.3.2020%20%2020_30.STEP

Haven't taken a look at them personally, more just pointing them out. :smile:

nimrod46, 2020-04-12:
If still have problems with the mechanic files feel free to reopen/open an issue

closed 2020-04-12 by nimrod46