AmboVent-1690-108/AmboVent #26: New Option PCB

Arie001, 2020-04-12:
Here is a new option  for a PCB 

https://github.com/Arie001/ambovent_pcb_Rev_Arie/tree/master/respirator-control_rev2

Please review Kicad  I have checked it in Fusion 360 .
I did not Find good cheap switches  that fit the small holes.
Only one three extra Parts for 12v .

DRC was run at 0.5mm .  Fails RGB Led the footprint  and Sw1 is near the corner screw
Note The Arduino and the Module I2C  was not put on the PCB as I want to keep it simple no smd soldering.

I think it is a Major step up. 
I can have a board ready in a few hours I am in Modiin  .

The  Board is supported 3 corners the bottom right is not supported No hole Below  the the Hole 
for the sensor.

I have exported the Step Not the Height of the Lcd may not be correct .

[image]

I use dry Film Etch  
054-6597555

salam17, 2020-04-12:
Not sure where this rant belongs, but here goes: Almost none of the DIY ventilator projects I see have alarms or battery back up, despite the fact that if you need a DIY ventilator you are also short staffed by definition; at a bare minimum, should power be lost when there are no eyes on the patient a  battery powered alarm is needed; it could save a life. Unless there is a UPS, one that has an alarm of its own, I see this as a critical requirement. There is plenty of room on this PC for a coin cell and a beeper.

nimrod46, 2020-04-12:
> ition; at a bare minimum, should power be lost when there are no eyes on the patient a battery powered alarm is needed; it could save a life. Unless there is a UPS, one that has an alarm of its own, [...]

That's correct we are using UPS  and a battery as a backup to **keep the system running** (not just for alarm). 

Someone did design for as a new board that will contain all of these things and **much more** but it will be ready only in a weak, @Arie001 I am contacting you NOW is will be a huge help for as.

Arie001, 2020-04-12:
See the PCB  I made see the PCB in the production directory . 
See the  gerber  f copper in the  github .I have spent time doing a high quality work .
No a 5 min pcb one that a person can be proud of.

Ok it was not me that claimed 2 hour battery life and never measured the battery voltage.If you claim 2 hours how do you know the battery was charged . Or was it Just that most of the Team wanted to get the positive press.   

I added 2 resistors and one Cap to a unused pin A7 .
Find Fault with the PCB it is High Volume ready done by a skilled Pcb Expert. Not somebody that does not Know the Difference between Front and Back Gerber Files.

I have no Issue to not have the PCB in the main.

Lets say you can  run this on a 12v 7ah  alarm led acid battery  for two hours . It is sort of Important 
to Know the battery is at 12v and not 10v.

Please download the original pcb a 1 layer board now explain how to solder the cap on the input if the board if the pads are covered. Fcu

Arie001, 2020-04-12:
Nimrod see the Board

Arie001, 2020-04-12:
it is about to be updated to esp32 in the next 2 days so memory issues will be history

Arie001, 2020-04-12:
Call Me  0546597555

Arie001, 2020-04-13:
I have Done Some extra Editing . To stop the Cables Falling out  the Headers face up just the Lcd  cable  needs securing  Hot glue  or silicon . I added a Tie down point
[image]

Arie001, 2020-04-15:
Here Is picture of my Current pcb My Fork has changed al lot But the Two parts that I can update the  Board Is the Pots and the Push Button switches> I am Moving away from the Idea of physical
Pots. 

I will Not we working on This  on the 16 of April  I an attending .
https://www.thethingsnetwork.org/conference/
 Online 
I am still TTN Modiin 
https://www.thethingsnetwork.org/community/modiin/

 
Arie
 
[image]

https://grabcad.com/library/ambovent-with-metalwork-face-1

Arie001, 2020-05-02:
Do not use until 3v level shift is added 2 * 2n7002 and 2 resistors

Arie001, 2020-05-06:
3v level shift done on pcb