AmboVent-1690-108/AmboVent #27: Resolve all warnings

ElectricRCAircraftGuy, 2020-04-12:
I just built the latest (from my PR #20), and there are dozens of warnings to fix.

```
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/arduino-builder -dump-prefs -logger=machine -hardware /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware -hardware /home/alexa/.arduino15/packages -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/tools-builder -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -tools /home/alexa/.arduino15/packages -built-in-libraries /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries -libraries /home/alexa/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path /tmp/arduino_build_472179 -warnings=all -build-cache /tmp/arduino_cache_729925 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -verbose /home/alexa/dev/AmboVent/AmboVent/3-Software/Arduino/ventilation_machine/ventilation_machine.ino
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/arduino-builder -compile -logger=machine -hardware /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware -hardware /home/alexa/.arduino15/packages -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/tools-builder -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -tools /home/alexa/.arduino15/packages -built-in-libraries /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries -libraries /home/alexa/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path /tmp/arduino_build_472179 -warnings=all -build-cache /tmp/arduino_cache_729925 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -verbose /home/alexa/dev/AmboVent/AmboVent/3-Software/Arduino/ventilation_machine/ventilation_machine.ino
Using board 'nano' from platform in folder: /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr
Detecting libraries used...
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries/Servo/src /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries/Servo/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/Wire/src /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries/Servo/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/Wire/src -I/home/alexa/Arduino/libraries/SparkFun_MS5803-14BA_Breakout_Arduino_Library-master/src /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries/Servo/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/Wire/src -I/home/alexa/Arduino/libraries/SparkFun_MS5803-14BA_Breakout_Arduino_Library-master/src -I/home/alexa/Arduino/libraries/Arduino-LiquidCrystal-I2C-library-master /tmp/arduino_build_47217 [... cut ...]
Using library Arduino-LiquidCrystal-I2C-library-master in folder: /home/alexa/Arduino/libraries/Arduino-LiquidCrystal-I2C-library-master (legacy)
Using library ArduinoUniqueID in folder: /home/alexa/Arduino/libraries/ArduinoUniqueID (legacy)
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_472179/ventilation_machine.ino.elf
Sketch uses 18412 bytes (59%) of program storage space. Maximum is 30720 bytes.
Global variables use 1046 bytes (51%) of dynamic memory, leaving 1002 bytes for local variables. Maximum is 2048 bytes.

```

nimrod46, 2020-04-13:
Fixed

ElectricRCAircraftGuy, 2020-04-14:
I just built your latest changes. 6 warnings still remain:
```

/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/arduino-builder -dump-prefs -logger=machine -hardware /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware -hardware /home/alexa/.arduino15/packages -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/tools-builder -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -tools /home/alexa/.arduino15/packages -built-in-libraries /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries -libraries /home/alexa/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path /tmp/arduino_build_472179 -warnings=all -build-cache /tmp/arduino_cache_729925 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -verbose /home/alexa/dev/AmboVent/AmboVent/3-Software/Arduino/ventilation_machine/ventilation_machine.ino
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/arduino-builder -compile -logger=machine -hardware /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware -hardware /home/alexa/.arduino15/packages -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/tools-builder -tools /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -tools /home/alexa/.arduino15/packages -built-in-libraries /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/libraries -libraries /home/alexa/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10808 -build-path /tmp/arduino_build_472179 -warnings=all -build-cache /tmp/arduino_cache_729925 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr -verbose /home/alexa/dev/AmboVent/AmboVent/3-Software/Arduino/ventilation_machine/ventilation_machine.ino
Using board 'nano' from platform in folder: /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr
Using core 'arduino' from platform in folder: /home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr
Detecting libraries used...
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src /tmp/arduino_build_472179/sketch/ventilation_machine.ino.cpp -o /dev/null
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/cores/arduino -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/variants/eightanaloginputs -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/arduino/avr/libraries/EEPROM/src -I/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/librari [... cut ...]
Using library Arduino-LiquidCrystal-I2C-library-master in folder: /home/alexa/Arduino/libraries/Arduino-LiquidCrystal-I2C-library-master (legacy)
Using library ArduinoUniqueID in folder: /home/alexa/Arduino/libraries/ArduinoUniqueID (legacy)
/home/alexa/Downloads/Install_files/Arduino/arduino-1.8.8/hardware/tools/avr/bin/avr-size -A /tmp/arduino_build_472179/ventilation_machine.ino.elf
Sketch uses 18412 bytes (59%) of program storage space. Maximum is 30720 bytes.
Global variables use 1046 bytes (51%) of dynamic memory, leaving 1002 bytes for local variables. Maximum is 2048 bytes.

```

ElectricRCAircraftGuy, 2020-04-14:
On medical (or any safety-critical) equipment, generally zero warnings are allowed with the compiler using its strictest settings.