AmboVent-1690-108/AmboVent #32: Code formatting: declare only 1 variable per line

ElectricRCAircraftGuy, 2020-04-13:
This is a sub-issue of #11. Don't close #11 until all sub-issues are resolved.

I will help with this; just want to teach the principle too is all. 

Also, I recommend the industry-standard `uint8_t` over the Arduino `byte` type. Byte is an alias to it anyway: "/home/alexa/dev/AmboVent/AmboVent/3-Software/Arduino/arduino_core/arduino/hardware/arduino/avr/cores/arduino/Arduino.h": line 126:

    typedef uint8_t byte;

In Eclipse, I just Ctrl + Click on any `byte` in the arduino source code and it jumps to its definition.

Instead of: 

    byte monitor_index = 0, BPM = 14, prev_BPM, in_wait, failure, send_beep, wanted_cycle_time,
         disconnected = 0, high_pressure_detected = 0, motion_failure = 0, sent_LCD, hold_breath,
         safety_pressure_detected;  

Do:

    uint8_t monitor_index = 0;
    uint8_t BPM = 14;
    uint8_t prev_BPM;
    uint8_t in_wait;
    uint8_t failure;
    uint8_t send_beep;
    uint8_t wanted_cycle_time;
    uint8_t disconnected = 0;
    uint8_t high_pressure_detected = 0;
    uint8_t motion_failure = 0;
    uint8_t sent_LCD;
    uint8_t hold_breath;
    uint8_t safety_pressure_detected;

Xexnon, 2020-04-13:
Yeah true the industry-standard uint8_t makes the code portable

ElectricRCAircraftGuy, 2020-04-13:
Related: #35

ElectricRCAircraftGuy, 2020-04-20:
PR #46 resolves a tiny portion of this issue