AmboVent-1690-108/AmboVent #38: Solid works version issue

labeneator, 2020-04-14:
I have a colleague who has solidworks 2019 installed but can't open the files due to a 'future version' error. This seems to be a [common issue ](https://forum.solidworks.com/community/faqs/blog/2017/12/22/why-am-i-getting-a-future-version-error-when-i-try-to-open-a-file) requiring one to use a later solid works. Is it possible to use an earlier version to save the file or use a vendor neutral file format?

Email from the said person in verbatim

---

> Thanks for the links to the CAD files for the trial project. I hope we can collaborate so we can try out any other designs for equipment that will alleviate the current medical equipment availability. 
> 
> I was able to access the zip file containing the Solidworks files. However, they are tagged as 'future version' meaning I can't open the attachments with my current Solidworks 2019.

---

Thanks

ElectricRCAircraftGuy, 2020-04-14:
I haven't used SolidWorks in 5+ years, so someone else chime in too please, but I'm pretty sure Solid Works goes to great lengths to ensure this is the case, and there may not be any work-around. When I was using a newer version of SolidWorks in 2013, there was absolutely no way to save it in an older format, specifically to force you to buy a license for the newer version so you don't have to. Therefore, a common strategy was to keep the same version for as long as possible between upgrades.

labeneator, 2020-04-14:
There seems to be a way to use vendor neutral files: 
 - https://forum.solidworks.com/thread/230295
 -  https://www.youtube.com/watch?v=FFrOpYIPtpk
 -  https://help.solidworks.com/2019/English/SolidWorks/sldworks/c_previous_release_interoperability.htm

However, you do seem to lose flexibility in editing the vendor neutral files. Perhaps we can have both file options published for better coverage.

nimamoslemy, 2020-05-01:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)