AmboVent-1690-108/AmboVent #41: Code formatting Variable name  index=0 line 138

Xexnon, 2020-04-14:
The line gave a compile time error while porting code to esp32.

Xexnon, 2020-04-14:
Related to #11 and #37

Arie001, 2020-04-21:
This was on platform.io  compiled for esp32 . Not checked on a clean install of platform.io I suspect 
one the name "Index" was   defined on one of the libraries  I have as global . It is not a bug it may be  
related to my setup. The reason for the compile

Is to try do the project on a more modern chip where memory is not a issue. 
 No More no Less not a bug or a Issue.

I my forked of the code 
ESP32  Rotary Encoder  Touch TFT  SD .
Wifi BLE  MQTT   

[image]