AmboVent-1690-108/AmboVent #52: Access to Materials

aligbenewton7, 2020-04-27:
Please, find attached a list of the materials I couldn't find around.
[Ambovent BOM (parts list)_check 3 sheets.xlsx](https://github.com/AmboVent-1690-108/AmboVent/files/4541308/Ambovent.BOM.parts.list._check.3.sheets.xlsx)

salam17, 2020-04-28:
In stock but pricier at Amazon: https://www.amazon.com/Dorman-742-600-Window-Lift-Motor/dp/B000E35WEC/ref=sr_1_1?crid=E9W6ABV6G81B&dchild=1&keywords=dorman+742-600&qid=1588098679&s=automotive&sprefix=dorman+742-%2Cautomotive%2C147&sr=1-1

aligbenewton7, 2020-04-28:
Greeting Team,

Please, where do I find these materials? I can't get them in Nigeria.

Part Quantity מספר יצרן Materials Measurements
0.5" HEX ID x 1.125" OD x 0313" 1 217-2734   VEXpro
Arm Left 1
Polycarbonate : t=6 180x70x6
Arm Right 1   Polycarbonate : t=6 180x70x6
Arm Standoff 3
ABS Printed DIA12x15
Inner Adapter 1   Polycarbonate : t=6 / ABS Printed DIA40x6
Hex Shaft 1
AL5052-H32 / AL6061-T6 HEX12.7x27.94
Hex Spacer 1   AL5052-H32 / AL6061-T6 DIA19.05x9
Poly Upper Base 1
Polycarbonate : t=6 345x264.12x6
Poly Buttom Base 1   Polycarbonate : t=6 345x264.12x6
Profile Base 2
Any Aluminium 25x25x2; L=264.12
Profile Engine 1   Any Aluminium 40x40x3; t=2/3
Wall Right 1
Polycarbonate : t=6 136x81x6
Wall Left 1   Polycarbonate : t=6 136x81x6
Spacer 1 2
AL5052-H32 / AL6061-T6 DIA12x40
Spacer 2 1   AL5052-H32 / AL6061-T6 DIA15.88x7.21
Adapter 1
ABS printed 46.6x30x22.3
Adapter Small 1   ABS printed DIA21x18
Top Plate 1
Polycarbonate : t=6 200x188x6
Buttom Plate 1   Polycarbonate : t=6 200x188x6
Spacer 4
AL5052-H32 / AL6061-T6 DIA12x50
Arm Left REV.2 1   Polycarbonate : t=6
Wall Left REV.2 1
Polycarbonate : t=6
Wall Right REV.2 1   Polycarbonate : t=6
Spacer 1
ABS Printed DIA29x3.65

Casing

Arie001, 2020-05-03:
This is Arie I am not part of the Ambovent Team  here is my Quick Google search  https://www.trilineng.com/about-us/   Triline Nigeria Limited   can create the Items From stock  Polycarbonate.

 You will need to supply them the files  the printed parts need a 3d printer.

https://www.finelib.com/business/printing-and-publishing/3d-printing-services

I think one can try a Automotive  Window motor . I do not Know if anybody has made a DXF for the whole Items Laser cut.