AmboVent-1690-108/AmboVent #53: Step by Step Guide?

salmanfarisvp, 2020-04-28:
Hi, This is not an issue.

Do we have any step by step by build guide for the AmboVent? I found the **AmboVent System Detailed Description - English 2.4.2020.pdf** , but it doesn't include basic things like instructions for the assembly..etc. 

Thanks.

Arie001, 2020-05-03:
Tell me what you need I hope to be in Bangalore soon after the covid19 ends . I do not think 
a step by step exists as only one prototype exists.  Even the BOM is a very gray area. Like analog feedback of motor position . How are they doing it is it a simple Pot or a proper sensor.For example
I was thinking of 3d Printed Sensor using a  Hall element OH49E SS49E Hall sensor  and a magnet .
sub less than a dollar or some IR method.

salmanfarisvp, 2020-05-18:
Did you completed the build?